Abdurraziq/ZMPT101B-arduino #2: Wrong Reading of Voltage

wahajmurtaza, 2019-07-08:
Hi, I am using your library to measure ac  voltage only  (modifying example code). I disconnected ac source during calibration as mentioned and even tried while being connected to ac source, I am getting values of more than 1000V while I have connected it to 220V AC source.

Abdurraziq, 2019-07-08:
If your connection is not problematic then the possibility is in the sensitivity setting. Have you tried adjusting the sensitivity of the sensor through the `setSensitivity()` method?
By default the value is 0.001. This value may not be the same between one sensor and another sensor. So please try with a different sensitivity value until the result is the same as a voltage source you use.

**example**

``` c++
#include "ZMPT101B.h"

ZMPT101B voltageSensor(A0);

void setup()
{
  Serial.begin(9600);
  Serial.println("Calibrating... Ensure that no current flows through the sensor at this moment");
  delay(100);
  voltageSensor.setSensitivity(0.019);
  voltageSensor.calibrate();
  Serial.println("Done!");
}

void loop()
{
  float U = voltageSensor.getVoltageAC();
  Serial.println(String("U = ") + U + " V");
  delay(1000);
}
```

wahajmurtaza, 2019-07-09:
I calibrated it using the setSensitivity() for 220v but when I switched to 150V it had some large inaccuracies.
I am working on the problem and will let you know.

Kingshahzad78, 2019-08-04:
Hi Dear,
I had alot of trouble in using ZMPT101b concerning the different coding I downloaded and libraries too. But when I downloaded library and coding you wrote. Every thing is fine and going well now.
Thanks for your efforts.

Bbadru, 2020-07-22:
Hi dear, i'm using your example and during calibration i was getting 5.3 volts but when i put on the supply i was getting 19v instead of 220 volts according to my multimeter

Bbadru, 2020-07-23:
i set my sensitivity to 0.002 and i got values close enough; but how can the sensitivity values be explained

Abdurraziq, 2022-06-16:
> i set my sensitivity to 0.002 and i got values close enough; but how can the sensitivity values be explained/ chosen. is it just random or is there a formula

In the simplest case you can use intuition 😁 by trying any value, and then start decreasing or increasing the value until you are sure that it is the right value.

But if you want to use a formula, I think the easiest method is to divide the value of the voltage you are measuring by the AC voltmeter between **_Vout_** and **_Vin_**.

But make sure that the AC voltmeter you are using has a good DC coupling capability so that only the AC voltage is measured in **_Vout_**. This is because the ZMPT module has a DC offset at its output so the zero point will shift approximately half of the VCC.

Sorry for taking too long to reply.

closed 2022-06-16 by Abdurraziq

Bbadru, 2022-06-17:
Thanks alot man

Abdurraziq, 2023-01-23:
> I never got an exact value, the avometer says 233v but your default code says 197v

Hi @BimoSora2  What board are you using? Unfortunately this library currently only supports 5V microcontrollers (eg Arduino UNO).

Have you tried this [suggestion](https://github.com/Abdurraziq/ZMPT101B-arduino/issues/2#issuecomment-509176140) before?
Here you need to determine the sensitivity, you can do it by trial error until the value is appropriate.