Abdurraziq/ZMPT101B-arduino #4: Aruidno halts with ZMPT101B

qamarhafeez, 2020-02-05:
Whenever there is a surge in electrical components like i turned on or off light of room or some other electric component, Arduino get halts when using ZMPT101B sensor. I am using this library to calculate voltage, please help.

Abdurraziq, 2020-02-05:
Hai @qamarhafeez, I can only suggest these two solutions;  first keep the electrical equipment away from Arduino and secondly use a stable power supply that has protection against the surge effect.  I think the Arduino is hanging/halts due to a power surge that affects the arduino clock so that the program that is supposed to run becomes a hang.

_I hope this will solve your problem._

qamarhafeez, 2020-02-06:
I have tested, the device halts only when AC power is added ZMPT101B sensor. Otherwise the device is working fine if AC power is disconnected from ZMPT101B. The device halts only when room switched is on or off.

qamarhafeez, 2020-02-06:
Can you please suggest me where to add stable power supply and also suggest me the power supply that is stable, i think 5volt power supply added to arduino is stable.

closed 2022-06-16 by Abdurraziq