Abdurraziq/ZMPT101B-arduino #9: incorrect reading using 120AC 60z

JimKruk, 2021-12-26:
I trying to following the demo using the ZMPT101B with an arduino and only trying to read AC line voltage.  I set the frequency on this line:  float U = voltageSensor.getVoltageAC(60); but when I run the code I see this in the serial monitor window:

Calibrating... Ensure that no current flows through the sensor at this moment
Done!
U = 13.59 V  (AC 120V plugged in)
U = 9.11 V  (unplugged AC 120V)

Here is the code:
`
#include "ZMPT101B.h"

ZMPT101B voltageSensor(A0);

void setup()
{
  Serial.begin(115200);
  Serial.println("Calibrating... Ensure that no current flows through the sensor at this moment");
  delay(100);
  voltageSensor.setSensitivity(0.019);
  voltageSensor.calibrate();
  Serial.println("Done!");
}

void loop()
{
  float U = voltageSensor.getVoltageAC(60);
  Serial.println(String("U = ") + U + " V");
  delay(1000);
}
`

here is a pic of the wiring:
[ 
[image]
](url)

thanks !

Abdurraziq, 2022-01-12:
Hi @JimKruk  did you input 120 V at the time of calibration?
Make sure that during calibration there is no input voltage

JimKruk, 2022-01-12:
Thank you for the reply
this is the program I am using from the example.  Are you saying there is a
parameter to specify 120v ? or to remove the line voltage when the
voltageSensor.calibrate(); line runs in the code during setup ?
thank you

 #include "ZMPT101B.h"

ZMPT101B voltageSensor(A0);

void setup()
{
  Serial.begin(115200);
  Serial.println("Calibrating... Ensure that no current flows through the
sensor at this moment");
  delay(100);
  voltageSensor.setSensitivity(0.019);
  voltageSensor.calibrate();
  Serial.println("Done!");
}

void loop()
{
  float U = voltageSensor.getVoltageAC(60);
  Serial.println(String("U = ") + U + " V");
  delay(1000);
}
runs ?  thank you

#include "ZMPT101B.h"

ZMPT101B voltageSensor(A0);

void setup()
{
  Serial.begin(115200);
  Serial.println("Calibrating... Ensure that no current flows through the
sensor at this moment");
  delay(100);
  voltageSensor.setSensitivity(0.019);
  C
  Serial.println("Done!");
}

void loop()
{
  float U = voltageSensor.getVoltageAC(60);
  Serial.println(String("U = ") + U + " V");
  delay(1000);
}

Abdurraziq, 2022-01-14:
Yes, you must remove the line voltage when voltageSensor.calibrate() line is running.
Sorry for my bad english,

JimKruk, 2022-01-14:
Thank you for your reply

closed 2022-06-16 by Abdurraziq