Abdurraziq/ZMPT101B-arduino #13: Not able to use with Arduino STM BluePill STM32F103C8T6 Board

asking23, 2023-06-22:
Getting error while trying to compile for STM Bluepill Board with arduino.

`:\Users\Documents\Arduino\libraries\ZMPT101B\src\ZMPT101B.cpp: In member function 'float ZMPT101B::getRmsVoltage(uint8_t)':
c:\Users\Documents\Arduino\libraries\ZMPT101B\src\ZMPT101B.cpp:60:55: error: 'ADC_SCALE' was not declared in this scope
   readingVoltage += sqrt(Vsum / measurements_count) / ADC_SCALE * VREF * sensitivity;
                                                       ^
c:\Users\Documents\Arduino\libraries\ZMPT101B\src\ZMPT101B.cpp:60:67: error: 'VREF' was not declared in this scope
   readingVoltage += sqrt(Vsum / measurements_count) / ADC_SCALE * VREF * sensitivity;`

Abdurraziq, 2023-06-27:
Sorry. Currently this library has not been tested on STM BluePill.
You may need to define the ADC_SCALE and VREF preprocessors before importing this library.

```cpp
#define ADC_SCALE 4096
#define VREF 3.3
#include <ZMPT101B.h>
```

asking23, 2023-06-28:
Ok defined in .C File of library and issue resolved.

closed 2023-06-28 by asking23